openwch/ch32v307 #2: Source code licensing

Tonymac32, 2022-01-08:
While the repo has a license, the individual source files do not, there is a copyright but no indication of how they can/shall be used:
```

/********************************** (C) COPYRIGHT  *******************************
* File Name          : core_riscv.h
* Author             : WCH
* Version            : V1.0.0
* Date               : 2021/06/06
* Description        : RISC-V Core Peripheral Access Layer Header File for CH32V30x
*******************************************************************************/
```

As an example, the header should contain what license the individual source file is distributed under to avoid confusion like this header from the Linux Kernel (clk-rk3288.c specifically)
```

// SPDX-License-Identifier: GPL-2.0-or-later
/*
 * Copyright (c) 2014 MundoReader S.L.
 * Author: Heiko Stuebner <heiko@sntech.de>
 */
```

HouHou01, 2022-01-10:
First of all, thank you for your feedback.Later versions will be revised.

vazhnov, 2022-01-26:
Hi!
Any update on this issue?
Opensource license is very important for a community.

HouHou01, 2022-01-26:
This update has fixed the issue.

closed 2022-02-09 by HouHou01