openwch/ch32v307 #6: Datasheet English translation

michele-perrone, 2022-01-22:
Hi! I was wondering if there are any plans to provide an English translation for the complete datasheet.  
I could find it only in Chinese: http://www.wch.cn/downloads/CH32FV2x_V3xRM_PDF.html

The datasheet that is currently available in this repository is the English version of [this document](http://www.wch.cn/downloads/CH32V20x_30xDS0_PDF.html), which does not include the detailed descriptions of each bit in the registers.

Thank you in advance!  
Michele

Tonymac32, 2022-01-22:
I second this, and thank you @michele-perrone for also linking the full (if untranslated) document

HouHou01, 2022-01-24:
First of all, thank you for your feedback。English version of the complete registers datasheet is being translated。When finished, it will be placed in this repository.

michele-perrone, 2022-01-27:
Thank you. This will be much welcome.

closed 2022-02-09 by HouHou01