openwch/ch32v307 #14: ETH examples do not seem to run

benpeoples, 2022-04-04:
I am utterly convinced I am just doing something wrong or missing a basic step.
 
Running MounRiver on Linux with the EVTv2 board.   

All of the examples I have tested work EXCEPT for the Ethernet ones.  I have tested every ETH project and none of them work, they all compile, but they print no information out the serial port (and the ETH port does not link).

Can someone confirm these examples work, or point me to where I might be going wrong?

HouHou01, 2022-04-04:
First of all, thank you for your feedback.The ram and ROM configuration of the WCHISPTool needs to match the ID file. 
The configuration is as RAMX 128KB + ROM 192KB.
[image]

_WCHISP Tool download link_
[](http://www.wch.cn/downloads/WCHISPTool_Setup_exe.html)

HouHou01, 2022-04-04:
WCHISP Tool download link。
http://www.wch.cn/downloads/WCHISPTool_Setup_exe.html

benpeoples, 2022-04-04:
Thank you! Is there any documentation of how this tool functions?   

Trying to avoid Windows components in my toolchain.  

(Not a dealbreaker, but would love to have all linux tooling here eventually)

benpeoples, 2022-04-05:
With updating the RAM/FLASH to 128K and 192K, the examples now run, thank you!

Donnyp751, 2022-04-27:
Any word on linux toolchain for the ch32v307?

benpeoples, 2022-04-27:
There has been discussions that one of the tools for another WCH microcontroller is easily ported (or possibly has been ported) to support the v307: https://discord.com/invite/2prGQmA3ez

closed 2022-04-27 by HouHou01

benpeoples, 2022-04-27:
I think the problem is that MRS does not support changing the RAM/ROM split, you have to use the WCHISP Tool which is only supported on Windows.  

Dug through the discord and found that folks seem to be talking about this tool: https://github.com/MarsTechHAN/ch552tool#usage

With PR#21 it supports v307 (reportedly, I have not tested), but I don't know if that includes support for adjusting the RAM/ROM split because I have not dug into it.  

Additionally, here's a couple of makefile-based projects that have been posted: https://github.com/gregdavill/CH32V307-makefile-example and https://github.com/infphyny/ch32v307_GPIO_Toggle