openwch/ch32v307 #16: Undocumented CSR in startup code: 0xBC0

karlp, 2022-04-28:
The CH32FV2x_V3xRM document has a special section for _Dedicated CSR registers_, but the provided startup code writes undescribed values to an undocumented CSR, 0xBC0
```
    li t0, 0x1f
    csrw 0xbc0, t0
```
See https://github.com/openwch/ch32v307/blob/main/EVT/EXAM/SRC/Startup/startup_ch32v30x_D8.S#L347-L348

What is this register for, and what do the values mean?  Riscv Privileged spec simply documents 0xBC0 as part of the range of custom machine read write registers.

This also applies to the CH58x parts, though there's a Chinese comment that implies this is to do with pipelining?

HouHou01, 2022-04-29:
First of all, thank you for your feedback.  0xBC0（csr）is used to configure MCU pipeline, instruction prediction and other related features, and users do not need to configure it.
Reference website of kernel related documents:
http://www.wch.cn/downloads/QingKeV4_Processor_Manual_PDF.html
ps：At present, the manual has only Chinese documents, and English documents have been uploaded recently

closed 2022-04-29 by HouHou01

karlp, 2022-04-29:
Ok, Well, I've not been setting this CSR, as I'm using my own startup code, and it ~works.  What am I missing out on?

karlp, 2022-04-29:
That's a great document though, already, thanks!

It refers to the Debugging protocol manual though? "具体的调试接口协议参考沁恒调试协议手册"  Is that available yet?  Having to use only the wch-link is a little awkward at times

HouHou01, 2022-04-29:
If 0xBC0（csr） is not configured, it will use the reset value, and the code will run normally, but it will affect the execution efficiency. Therefore, 0xBC0（csr） is recommended to configure this register to 0x1F