openwch/ch32v307 #19: UART server not building and not working

DualTachyon, 2022-05-19:
Hi,

I received 2x EVT boards recently and I tried to make the UART server run on one of them. However I'm unsuccessful so far.

The first problem is with the RT-Threads version of the server. I installed the latest version of Studio and it will not compile due to conflicts between the GCC toolchain (RISC-V WCH version) and system headers (from RT Threads). I didn't want to find such conflicts since it involved changing global headers, so I tried the baremetal version.

Baremetal version builds successfully and I can flash it to the EVT without issue. However the server does not appear to initialise the Ethernet port successfully. I confirmed with the "Debug" option of MounRiver that it passes the PHY_LINK_SUCCESS test. However, no LED is turned on on the Ethernet port. I correctly modified the IP address settings to match my network (192.168.0.x).

I ran different samples from the EVT folder as well as the USB HID+CDC demo and they are working fine, so I don't think it's a general board issue.

jpliew, 2022-05-20:
Have you connect the LED to the right GPIO ? I don't have the board with me now, I think it is called ELED1 and ELED2. Check which GPIO is init from the source code. Also check if you are on the latest MounRiver v1.7 

The example should work, but very unstable.

jpliew, 2022-05-20:
One more thing, have you change memory setting to  128KB RAM + 192KB Flash ?

closed 2022-06-10 by HouHou01