openwch/ch32v307 #27: Is it really open source? Where can I found the source code of "libwchnet.a"

gerryqd, 2022-07-06:
As title.

Nable80, 2022-07-06:
+1, although I'm much more interested in the source code of network stack for CH579, which is based on a modified version of lwip.
Unfortunately WCH refused to provided it, so I had to postpone using their chips in my devices (and continue using clones of STM32F107/F407 from other vendors) until I implement my port of lwip for this MCU. WCH's hardware is awesome but quality of their libraries (and values of compile-time options) doesn't suit me. I still hope to use them in the future.

closed 2022-07-07 by HouHou01

Nable80, 2022-07-07:
@HouHou01 请您提供任何意见好吗？

gerryqd, 2022-07-16:
@HouHou01 
It is quite funny that closed my issue without any comments.

fdarling, 2022-07-18:
I too have the same question... is the WCHNET library actually open source? It seems that the example code uses this opaque binary static library:

https://github.com/openwch/ch32v307/blob/main/Application/8_uarts_server/bare_metal/8_uart_server/ETH/User/libwchnet.a

Only the API C header file is available, not the actual implementation:

https://github.com/openwch/ch32v307/blob/main/Application/8_uarts_server/bare_metal/8_uart_server/ETH/User/WCHNET.h

You can see that the binary library is definitely being linked into the example:

https://github.com/openwch/ch32v307/blob/main/Application/8_uarts_server/bare_metal/8_uart_server/ETH/obj/8_uart_server.map

Please re-open this issue and let us know what's going on!

HouHou01, 2022-07-22:
您好，libwchnet.a为我公司私有的网络协议栈库，该协议栈库不开源，在使用C32V307的ETH做开发时，建议使用我方私有协议栈进行开发（具体操作可参考WCHNET Protocol Stack Library Application Note.PDF），开发者也可使用第三方开源协议栈。

Nable80, 2022-07-22:
@HouHou01 谢谢您的回复！
将此代码保密的确切原因是什么？ 无意冒犯 （我不想冒犯任何人），我只是想了解原因。WCH 担心竞争对手吗？

我知道开发代码需要时间和精力，但其他供应商正在发布他们的库。 公共代码对竞争对手没有帮助，但对客户有帮助。

我祈望这条消息是可读的。如果我错了，请纠正我。

> 开发者也可使用第三方开源协议栈。

是的，我选择了这种方式。

---
Thank you for your reply!
What is the exact reason for keeping this code private? No offense, I just wanted to understand the reasons. Is WCH worried about competitors? I know it took time and effort to develop this code, but other vendors are releasing their libraries and it's not a problem. Public code doesn't help competitors and it helps customers.

I hope my message was readable. Please correct me if I wrote anything wrong.

> 开发者也可使用第三方开源协议栈。

Yes, I'm going this way. Thank you for information.

maximmeshkov, 2023-08-14:
it is extremely sad that the library appears only in such a not digestible form

Llahainur, 2023-08-16:
Im having some problems with using wchnet library with FreeRTOS. Maybe some interrupts were changed in the library and it's not compatible?