openwch/ch32v307 #28: issue to Debug CH32V307V with Visual Studio Code on macOS because wch-link version 2.4(I need v2.3)

neuberfran, 2022-07-11:
Hi,

I stay with this issue in StackOverflow in last 10 days:
https://stackoverflow.com/questions/72863670/failed-to-attach-3333-operation-timed-out-from-target-select-remote-3333

**The issue appear here:**
https://gist.github.com/neuberfran/ab46bab4a388b03255d851cff1cfd537#file-issue-das-issue-ch32v307-no-mac-os-catalina-big-sur-L12

I now the soluction is use wch-link V2.3(Not 2.4). But I don't know How to downgrade or How to compile directly to version 2.3(Despite having tried, as can be seen in the link discussions below with the author of the tutorial):
https://jia.je/hardware/2022/04/19/wch-ch32v307-eval/

How can I solve this?

Pls

HouHou01, 2022-07-12:
HI,
Updata description file URL:https://github.com/HouHou01/Link-Bin/tree/main/doc.
V2.3 fireware URL:https://github.com/HouHou01/Link-Bin/tree/main/code/V2.3
ISP Tool URL:https://github.com/HouHou01/Link-Bin/tree/main/ISP_Tool

closed 2022-07-12 by HouHou01

neuberfran, 2022-07-12:
@HouHou01 I need step by step in English.

How to I will use WCHLINK_V02.03.bin file to downgrade wch-link version 2.4 to 2.3 and solve my StackOverFlow ISSUE?

[image]

Pls