openwch/ch32v307 #29: Trying to figure out memory config

udif, 2022-07-17:
The CH32V307 has (according to the datasheet) , 256KB Flash memory and 64K RAM.
Yet, the datasheet mentions the following:

> The products with 256K FLASH+64K SRAM support user select word to be configured as one of several combinations of (192K FLASH+128K SRAM), (224K FLASH+96K SRAM), (256K FLASH+64K SRAM), (288K FLASH+32K SRAM)
You also see this when you select Flash/Configuration in MRS, where you can select the memory configuration.

It is still not clear to me. Flash memory is nonvolatile, and physically different from RAM. How can you configure the same chip to have different Flash/RAM configurations on the same silicon?

jpliew, 2022-07-21:
The CH32V307 has a total of 320KB SRAM and 488KB FLASH (undocumented)

The 320KB RAM is divided into 2 parts. One part is used as RAM and the other part is used for CODE (Some part of flash is copied into RAM to speed up code execution)

For example, if you configure

192K FLASH + 128K RAM = TOTAL 320 KB RAM

The RAM starts from 0x2000000 ~ (0x2000000 + 128K)
The FLASH starts from end of the RAM (0x2000000 + 128KB)

When 192KB FLASH has been copied to the RAM, the balance of 296KB (488 - 192 ) can be used as storage.

Unfortunately, the datasheet is not very descriptive about this. Hope this helps.

JP

HouHou01, 2022-07-22:
Note: or the purpose of copying from FLASH to RAM, the system clock can break through the limitation of FLASH.

closed 2022-07-22 by HouHou01