openwch/ch32v307 #32: Fail to convert data hex file to bin

berkutta, 2022-08-05:
Download from MounRiver Studio seems to be not working. Download via WCHISPTool works fine.

```
------------ Begin flash process of "obj\DHCP.hex" ------------ 
20:24:14:344 >> Current project vendor is WCH, debugger is WCH-Link

20:24:14:344 >> Attempt to open device and upgrade firmware if necessary...
20:24:14:358 >> Fail to convert data hex file to bin

-----------------------------End ------------------------------
```

HouHou01, 2022-08-08:
Hello,Mrs needs to be updated to V1.81.
Update package download link-
http://www.mounriver.com/download

closed 2022-08-08 by HouHou01

berkutta, 2022-08-08:
@HouHou01 Doesn't work with V 1.81

[image]

berkutta, 2022-08-08:
It works with WCH-Link but not with Bootloader

HouHou01, 2022-08-09:
[image]
@berkutta
1..Please check whether there is WCH link in the device manager.
2.Please check whether the firmware version of the WCH-Link is correct.
MRS(V1.81)——WCH-Link（REV_0205）

HouHou01, 2022-08-09:
Try to WCh-Link update the v2.5 firmware.
Refer to https://github.com/HouHou01/Link-Bin