openwch/ch32v307 #45: IAP APP interrupt fail

slimelec, 2023-02-05:
After jumping to new APP and enabling interrupts the USB stops working.

Usually in ARM we load the new APP interrupt vector after the jump. Is there anything equivalent for CH32 ?

qiyongshuang, 2023-02-06:
The APP project is downloaded through the IAP project, the IAP plays the role of the burning program, and the APP plays the role of the MCU function. The configuration of the USB register in the IAP is only to configure the download method of the APP, and the USB register is cleared after downloading. If you want to operate the MCU registers, you need to configure them in the APP. For IAP User Manual, you can refer to： https://github.com/qiyongshuang/IAP,
I hope it can solve your problem. If you still have problems, you can share the program with your permission.

slimelec, 2023-02-07:
@qiyongshuang Here's how to replicate the issue. In this example I'm using "EVT\EXAM\IAP\USB_UART". The "CHV30x_IAP.hex" was built with no modification to the code. 

The "CHV30x_APP.bin" was flashed using the supplied "WCHMcuIAP_WinAPP.exe",  And I made small modifications to CHV30x_APP code. You can see the output below, "loop" is never printed. 

At first I thought it might be interrupts, but turns out it's the SysTick which is the issue.

Here is the test projects: [USB_UART.zip](https://github.com/openwch/ch32v307/files/10674238/USB_UART.zip)
Here is the schematics for the dev board I'm using:  [YD-CH32V307VCT6-SCH.pdf](https://github.com/openwch/ch32v307/files/10674261/YD-CH32V307VCT6-SCH.pdf)

```
int main(void)
{
    u16 i;

    SystemCoreClockUpdate();
    Delay_Init();
    USART_Printf_Init(115200);	
    printf("SystemClk:%d\r\n",SystemCoreClock);
    printf( "ChipID:%08x\r\n", DBGMCU_GetCHIPID() );
    printf("APP\r\n");

    ADC_Function_Init();
    printf("CalibrattionValue:%d\n", Calibrattion_Val);

    DMA_Tx_Init( DMA1_Channel1, (u32)&ADC1->RDATAR, (u32)TxBuf, 1024 );
    DMA_Cmd( DMA1_Channel1, ENABLE );

    ADC_RegularChannelConfig(ADC1, ADC_Channel_2, 1, ADC_SampleTime_239Cycles5 );
    ADC_SoftwareStartConvCmd(ADC1, ENABLE);
    Delay_Ms(50);
    ADC_SoftwareStartConvCmd(ADC1, DISABLE);

    for(i=0; i<10; i++)	//changed from 1024 to 10
    {
        printf( "%04d\r\n", Get_ConversionVal(TxBuf[i]));
        Delay_Ms(10);
    }
    printf("adc print done.\r\n");    // Mod
	while(1)
	{
		Delay_Ms(1000);    // Mod
		printf("loop");        // Mod
	}
}
```

output:
[image]

qiyongshuang, 2023-02-08:
This is a matter of flushing the cache. You can add \r\n or choose our library (without adding \r\n).
[image]
[image]

slimelec, 2023-02-08:
I don't have this option in my settings. How can I get this library?

qiyongshuang, 2023-02-09:
MRS Community Edition does not have this option, for your convenience to use MRS.You can re-download an official version of MRS from the official website.http://www.mounriver.com/download