openwch/ch32v307 #46: Sharing code between MounRiver projects.  Suggestions?

KurtE, 2023-02-05:
Sorry, I know this is probably not the best place to ask this, and the answer is probably already here, I just have not found it.

But thought I would ask. 

A few if us our experimenting with these chips and seeing how hard it would be to create a set of logical libraries and to then have several test programs that use some or all of these libraries. 

Short version: Is there some form of project type/template that does not build an executable, but instead creates a library (*.a) file?
I am assuming there are a bunch of manual steps one could do now to make one? 

And then is there an an easy way to create one or more projects within the solution that, references these libraries?  I am assuming that I would probably need to change the includes for  c/c++/link to reference these.

Right now, I am doing sort of a kludge.   I create the logical library as a top-level directory where my workspace(solution) is.
I then go into each project that wants to use this code, and right click on the top-level directory for that project and select new folder.
like:
[image]
issue 1: I am guessing though that the link is kept with my path and not relative to the workspace.

Then I go into the project and update the includes to include these directories using the logical link directories, like:
[image]

This is sort of working, but looking at the verbose link, it appears like it directly references each generated .o file , so it probably includes lots of code and data usage that is not needed. 

Example suppose, I have a set of library objects for each of the possible 8 USART/UARTs including RX and TX buffers and the like.  But my sketch only uses 1 of these.  How to I keep the executable from bringing in all of those unused resources.

one option, which I have and do use, is to have a config file or the like, that has #includes for each of the used resources.  Which works.  But...

In some other projects, I have tried to make it sort of Automatic.  

That is suppose that everything specific to a specific Serial object is self-contained within a source file.  That is, I have separate file for Serial1, Serial2, Serial3...  And no other source files within the library directly references anything within these files.  The files are compiled as part of a library which generates an archive (.a) file.  So only if something external from this library references this object, will the linker include the stuff within these files.

So wondering if this makes sense here, and hopefully is easily doable?

Thanks
Kurt

qiyongshuang, 2023-02-07:
Regarding your issue 1, if the file is not in the workspace, you need to click "file systrm" to add the path. for what you said "includes lots of code and data usage that is not needed"  I think you can refer to “CH32V30xEVT\SRC\Debug.c”.Add macro definition to reduce useless code.
Hope this solves your problem, if you still have problems you can share the program with your permission. for a more detailed analysis.

KurtE, 2023-02-07:
Thanks, looks like a good place to start.

KurtE, 2023-02-28:
Thanks, I will go ahead and close this one.

Only thing I am trying to figure out next is if there is a way for the Solution/Workspace to have the main set of settings and the like, that each of the contained projects can use by default...

closed 2023-02-28 by KurtE