openwch/ch32v307 #49: Schematic/Board information - For Ebay/Amazon CH32V307WCH breakout board.

KurtE, 2023-02-13:
Wondering if you have information on the breakout board(s), that I purchased one from Ebay and the 2nd from Amazon:
https://www.ebay.com/itm/225184379780
https://www.amazon.com/dp/B0B7TJ5F7V
[image]
[image]

Note: the product pages do give some information, like what IO pins the LED/LED/User button is on,

But no information on for example what the different jumpers on the back do.  

Likewise, I tried the one SDIO example and it did not appear to want to talk to either of the two SD cards I tried in them.  But maybe this is controlled by one or more jumpers. 
C8-C12, D2?  maybe some of those jumpers?

There is not much information on how the flash or EEPROM is hooked up, but probably can deduce or ping out.

I have not checked out the Ethernet yet, but I am assuming that the RXN/RXP, TXN/TXP are used for it. 
Probably pins C6-C9? 

The Amazon listing does have an image of a Schematic, but the image is so small and fuzzy, that I can not  read it.

So again, wondering if you might have additional information on these boards, like you have in the SCHPCB directory of this project for your Rev0 and Rev1 boards.   I have ordered a Rev1 board, which has shipped and is due to arrive next month. 

Thanks
Kurt

qiyongshuang, 2023-02-14:
Sorry, this is not our product. So I have no relevant information. CH32V307VCT6 chip pin information, you can refer to http://www.wch-ic.com/downloads/CH32FV2x_V3xRM_PDF.html  You can purchase WCH products through https://www.aliexpress.com/store/1102052740/search?spm=a2g0o.store_pc_allProduct.8148362.6.2c7439e4rjo1Q8&origin=n&SortType=orders_desc.

KurtE, 2023-02-14:
Thanks, thought it was a long shot.  That maybe there was some additional info. 

As I mentioned I have one of your 307V1 boards ordered (through Amazon) that has shipped should be here within the next month.

closed 2023-02-14 by KurtE

KurtE, 2023-02-15:
Quick update, in case anyone is interested.  A friend who also working with these received a schematic which he sent me. 

And it helped me confirm, that the SDIO socket on the board will not work without changing some/all of the Solder jumpers on the back of the board. 

That is, it appears like a board can Either use SDIO or Ethernet, but not both, as they both use pins PC8 and PC9, which you can see in this section of the Schematic:
[image]
And I don't see any alternative pins setup to change them.   

Now back to playing