openwch/ch32v307 #50: Issue with remapping TIM9

techlabz, 2023-02-13:
Chip: CH32V307VCT6

I am trying to remap TIM9 and it is not working. I did Full Remap to use TIM9_CH2 on PD11, in default configuration(PA3) it works fine. But doesn't seem to work in Full Remap or Partial Remap.

[image]

qiyongshuang, 2023-02-14:
Thank you for choosing our product. I remapped TIM9 and the function is normal. Attach the main procedure and results. If you have any questions, you can share the program with your permission.
[image]

qiyongshuang, 2023-02-14:
Thank you for choosing our product. I remapped TIM9 and the function is normal. Attach the main procedure and results. If you have any questions, you can share the program with your permission.
[image]
[image]
[image]
[image]

techlabz, 2023-02-14:
Hi @qiyongshuang, made silly error, used TIM_OC1Init instead of TIM_OC2Init, thanks again.

closed 2023-02-14 by techlabz