openwch/ch32v307 #52: SPI - NSS pin - Chip select - hardware, versus Software versus just using GPIO.

KurtE, 2023-03-04:
Sorry it is probably just me, but it is unclear to me on the proper way to control the CS for one or more devices that are connected to an SPI buss.  In my case Master mode.

Especially since I may want to interact with more than one device using SPI on a single buss (and you only have one active NSS pin), 
I typically would control the CS pins by software as to control which device the SPI transaction is associated with. 

So my assumption is to set the SSM  bit in the SPI1.CTLR1 register to high. 

However suppose in this case I happen to have an SPI device connected up and in this case it's CS pin is connected to PA4 which is the NSS pin.  

It is unclear on what the usage of the SSI bit is: 
```
CS pin control. When SSM is set, this bit
determines the level of the NSS pin.
1: NSS is high level;
0: NSS is low
```
So is this implying that I must use this the SSI to control the state of this pin, but for all other CS pins, I would use GPIO to set/clear?

The defines in the SDK don't fully match the Reference manual:
```
#define SPI_CTLR1_SSI                           ((uint16_t)0x0100) /* Internal slave select */
#define SPI_CTLR1_SSM                           ((uint16_t)0x0200) /* Software slave management */
```

One reason I am asking this, is for example the first CH32V307 board I purchased has a flash memory chip on board with it's CS pin set to use the PA4.  

But I am also asking this, to better understand that if I should use this SPI buss or one of the others.  Is the NSS pin always taken over by SPI, even if I have software set and as such I can not use this pin for some other purpose? 

Thanks
Kurt

qiyongshuang, 2023-03-07:
When SPI is used as the host, SSM is set to high and SSI is set to high, and the internal pin of NSS is set to high. NSS external pins can be used for other purposes. At this time, the NSS external pin can be used as a chip select  pin.

KurtE, 2023-03-14:
Thanks for the information.

The only thing I found confusing, is understanding what the differences are between the internal pin of NSS and the NSS external pin.  

But things appear to be working, so again thanks

closed 2023-03-14 by KurtE