openwch/ch32v307 #57: How to check Flash and SRAM usage?

techlabz, 2023-03-11:
Is there a way to check the size of Flash and SRAM usage? I can't seem to find either in the wch download utility tool and mounriver studio, I can see the hex file, but not the actual binary size.

qiyongshuang, 2023-03-13:
You can generate BIN files through mounriver studio.
Click "file"->choise "properties"
[image]

techlabz, 2023-03-15:
Hi @qiyongshuang,

What about SRAM/Memory usage?

techlabz, 2023-03-17:
@qiyongshuang  any luck?

qiyongshuang, 2023-03-19:
After the project is compiled by MRS:
[image]
flash=data+dec;
rom=data+bss;

qiyongshuang, 2023-03-19:
After the project is compiled by MRS:
[image]
flash=data+dec;
SRAM=data+bss;

closed 2024-01-02 by qiyongshuang