openwch/ch32v307 #59: Does ch32v307 really support isochronous FS/HS transfer types?

nikkov, 2023-03-27:
I didn't find any samples with this type of USB transfer, so I don't know if it's possible to make an USB-audio device on this chip or not.

techlabz, 2023-04-05:
I have been working on this for a few days, and I can't get isochronous connection to start. Any examples would be helpful.

techlabz, 2023-04-05:
I figured out the issue, it looks like there may be a bug with USB 2.0 High Speed, it works with full speed. Technically speaking with USB Audio Class 1.0 does only need Full-Speed to work. Maybe there may be a reason it does not work with High Speed not sure.

[image]

closed 2024-01-02 by qiyongshuang