openwch/ch32v307 #60: RTT implementation CH32V307

sadkotheguest, 2023-03-28:
Hi! 
I'm trying to implement RTT (real-time-transfer) feature for debug and tracing of code with OpenOCD.

Wiki of Segger said that
_RTT background memory accesses are performed either via via RISC-V system bus access (SBA) or via AHB/AXI-AP_.
( https://wiki.segger.com/RTT#RISC-V_specifics )

Could you tell - are SBA and/or AHB/AXI-AP access available in CH32V307 core?

martinribelotta, 2023-09-24:
Some progress here? I'm working on it but need to enable/disable interrupt in nested way.

i found this in `EVT/EXAM/SRC/Core/core_riscv.h`:

```c
__attribute__( ( always_inline ) ) RV_STATIC_INLINE void __enable_irq()
{
  __asm volatile ("csrw 0x800, %0" : : "r" (0x6088) );
}

__attribute__( ( always_inline ) ) RV_STATIC_INLINE void __disable_irq()
{
  __asm volatile ("csrw 0x800, %0" : : "r" (0x6000) );
}
```

But the documentation not said anything about CSR 0x800 (non standard extension)

I found this in the documentation (QingKeV4_Processor_Manual.PDF) but the excact data field is not mentioned:
[image]

If I read bits [15:0] of CSR 0x800 and restore it I can simulate multiple nested `__disable_irq()/__enable_irq()`? What others bitfield in CSR 0x800 is dangerous?

martinribelotta, 2023-09-24:
Some updates:

 - `ginter` is effectively 0x800
 - Aparently, it is a shadow copy of mstatus in bits 3 and bit 7 (MPIE:7, MIE:3)
 - This **aparently** disable/enable interrupts correct from user mode

The big problem now is enable RTT on openocd.

You can see the progress in my repository (add-rtt branch): https://github.com/martinribelotta/openwch-makefile/tree/add-rtt

martinribelotta, 2023-09-27:
A patched wch openocd with rtt risc-v thread support can be found here: https://github.com/martinribelotta/openocd-wch/tree/rtt-riscv-wch-support

VSCode working with cortex-debug RTT support:
[image]