openwch/ch32v307 #87: SDK V2.4: new test in IWDG_Enable() in ch32v30x_iwdg.c looks incorrect

OpusElectronics, 2024-04-02:
Hi, I noticed that a new test has been added in the IWDG_Enable() function in ch32v30x_iwdg.c .
This new test is the following line:
```while((RCC->RSTSCKR)|(0x2)!=SET);```

I think it's incorrect and can only lead to an infinite loop. I'm assuming it was meant to test the LSIRDY bit of the RCC_RSTSCKR register? But here, by or'ing it with 0x2, it will never be equal to 'SET'.

If the intent was to wait until the LSIRDY is set, the correct way would be (that's how I've modified it in my local repo):
```while (! (RCC->RSTSCKR & RCC_LSIRDY)) {}```

(Note that the RCC_LSIRDY is already defined in ch32v30x.h, so we may as well use it instead of a hard-coded value. Also, small detail, note that I prefer using {} as an empty statement as while/for loops bodies, it's easier to read and less error-prone than single ';' - but that's a detail here.)

qiyongshuang, 2024-04-03:
You are right ,our firend.
We are so sorry ,We will fix it in the next version.

OpusElectronics, 2024-04-03:
No problem, happy to help!

closed 2024-05-09 by qiyongshuang