openwch/ch32v307 #92: Filename too long when checking out repo in Windows

paulhayes, 2024-07-20:
When checking out in Windows, as part of a submodule in usbtiny. I got the following error.

```
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0001-RTT-VMM-implement-dual-system-running-on-realview-pb.patch: Filename too long
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0002-arm-gic-correct-the-cpu-map-on-gic_raise_softirq-for.patch: Filename too long
```
To fix it I had to run the following as administrator

```
git config --system core.longpaths true
```

and in Powershell

```
New-ItemProperty -Path "HKLM:\SYSTEM\CurrentControlSet\Control\FileSystem" `
-Name "LongPathsEnabled" -Value 1 -PropertyType DWORD -Force
```

qiyongshuang, 2024-10-10:
EVT file have rt_thread routine , you can try it.

paulhayes, 2024-10-10:
I'm sorry, I don't understand. The issue I reported is due to filename that is too long for Windows 10 to support without being adjusted across the whole system.

closed 2024-11-11 by qiyongshuang